OctoPrint/plugins.octoprint.org #1125: [Abandoned Plugin] OctoPrint-SystemCommandEditor

synman, 2022-12-01:
### Which Plugin?

https://plugins.octoprint.org/plugins/systemcommandeditor/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/Salandora/OctoPrint-SystemCommandEditor/issues/27

### Is there someone open to adopting it?

@synman

synman, 2022-12-01:
More than happy to take this over.   I have a fork setup and am planning on changing the WYSIWIG editor style @Salandora implemented to a more straightforward approach, similar to how @jneilliii manages events in the (built-in) Event Manager plugin.

jneilliii, 2022-12-01:
Thanks @synman, after 14 days if we have no response from @Salandora you can submit a PR for the adoption process.

closed 2022-12-26 by jneilliii