OctoPrint/plugins.octoprint.org #1126: [Abandoned Plugin] TouchUI

roblandry, 2022-12-01:
### Which Plugin?

https://plugins.octoprint.org/plugins/touchui/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/535

wwian, 2022-12-02:
<sad face. I still use this app with my prints.>

jose-campos, 2022-12-04:
What’s the process to take this over? Fork and re-submit?

jneilliii, 2022-12-04:
Fork, address any outstanding major issues relative to compatibility, etc. bump version and submit PR to redirect the update checks. A good example can be found [here](https://github.com/OctoPrint/plugins.octoprint.org/pull/1041/files) of such a PR.

jose-campos, 2022-12-04:
Sounds good, will install and check the issues to see how much work may be involved and report back.

vexofp, 2022-12-04:
Probably the biggest issue is an incompatibility with Octoprint 1.8.0+. See https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/532 for more info.

There is a link in there to a branch I made and have been using locally that fixed that problem, but I was never entirely happy with it because it broke backwards compatibility with older Octoprint versions. Maybe that is less of a problem now, since 1.8.0 has been out for awhile now? Either way, hopefully it helps!

JohnathonMohr, 2022-12-06:
I've been wondering what happened to the temp graph. 😅

Well, if the plugin transfers ownership, that would provide a convenient time to shift compatibility to 1.8.0+. Just my 2 cents.

sfarkhan, 2022-12-28:
I use TouchUI as well. I would be up to Patreon someone to take over. It's a popular plugin

TheNore, 2023-02-16:
Has there been any update on this? I use this on all of my printers, I would gladly donate to whoever took this over, its a great Plugin!! It would be ashamed to loose it.

jneilliii, 2023-02-17:
> but I was never entirely happy with it because it broke backwards compatibility with older Octoprint versions

You could do a copy of the file and conditionally load the different file based on OctoPrint version. Here's an example of where I've used this.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/27d265aadf164f2f2aebad99ef64874af029d869/octoprint_tplinksmartplug/__init__.py#L337

Also, it's always better to utilize id css selector if at all possible, just not sure without looking directly at it if that's available here or not.

Thierry-Myron, 2023-02-21:
So sad... Still using on 10.1 touchscreen :(

Unn4m3DD, 2023-03-01:
I'm considering taking a look at the plugin, fixing the graph, and maybe some quality-of-life updates, just asking here if someone else is already on it so that we don't duplicate our efforts

Unn4m3DD, 2023-03-01:
Well, I have it sorted out. I've created a fork with the changes, can someone help me with how to make the update reach octoprints repositories? https://github.com/Unn4m3DD/OctoPrint-TouchUI
[image]

vexofp, 2023-03-05:
@Unn4m3DD , I think the legend and temperature values at the bottom there still look messed up. They should look like this:
[image]

Please feel free to steal from https://github.com/BillyBlaze/OctoPrint-TouchUI/compare/master...vexofp:532-temp-graph-1.8.x (if helpful)!

Unn4m3DD, 2023-03-08:
I thought those bars were sliders xD I managed to get to this, should be close enough
[image]

Unn4m3DD, 2023-03-17:
Heyy, I have had the PR opened for some time and I'm wondering if I did it in the correct way. @vexofp, you seem to know your way around the octoprint community, any chance you might be able to help me understand this?

jneilliii, 2023-03-18:
If you want to adopt the plugin it's just a matter of submitting a PR here that configures the correct update check overlays, the plugin listing data adding you as an author and a bumped version with release on your fork of the plugin. Here is an example of a completed adoption PR to see what I mean.

https://github.com/OctoPrint/plugins.octoprint.org/pull/1157/files

ThatMattCraig, 2023-04-03:
I'll take this over. Ill try my best to keep it running for now

pzwe2020, 2023-05-11:
My friend and i would love to transform this project and fix it. We will work our hardest on keeping it up. We just need someone to convert everything to python! I am new to get hub, and need to know how to take over the project.

pzwe2020, 2023-05-12:
*Github

Mbayazeed, 2023-06-16:
can't get a cam stream either !!

devildant, 2023-06-22:
Good morning,
unfortunately I don't have time to resume this plugin, however I made a fix for the webcam on my fork if you want:
https://github.com/devildant/OctoPrint-TouchUI

just download the master branch as zip, and install manually inside octoprint

devildant, 2023-06-23:
Note: I applied the fix from @Unn4m3DD + fix the SD card symbol in the files section

devildant, 2023-06-23:
[image]

[image]

[image]

Mbayazeed, 2023-06-25:
That worked perfectly.

Thanks

Kindest regards .

Sent from BlueMail<https://bluemail.me>

https://github.com/devildant/OctoPrint-TouchUI

chuwy5, 2023-08-05:
is someone maintaining it? if not depending on the code I might be willing to tip my hat in to atempt to keep it going

devildant, 2023-08-05:
> is someone maintaining it? if not depending on the code I might be willing to tip my hat in to atempt to keep it going

at the moment there is no one to take over, i just made patches on my branch to fix the problems on the latest version of octoprint. if you want to resume the project, go go :), it will please a lot of people

pzwe2020, 2023-08-05:
I have been working on trying to figure out the code and I just can get it. Hopefully we can find someone else. Sorry

chuwy5, 2023-08-06:
no worries. my thought is to find a group of people that can work on it.
unless i can figure it out and start maintaining it. but im down to work on
it with others

JayroJones, 2023-09-02:
Jesus fucking christ, the plugin STILL works fine, and I'm tired of the UI nagging me to remove it because it's "abandoned" by it's creator. (Thanks for that, _douche_.) How do I get it to stop nagging my system, even-though I always hit "Mark as read."? The nag is annoying as fuck to deal with, but the plugin is too useful to get rid of.

cp2004, 2023-09-02:
@JayroJones - I suggest you choose a different tone so that you come across as a reasonable human being. Please check https://octoprint.org/conduct/.

Marking as read sets a cookie so the notification doesn't show again. If you're having trouble with this, I'd suggest politely trying to seek help on the community forums with appropriate gratitude for anyone who now decides to help you, and remember first impressions count.

K1Printing, 2023-12-15:
If someone adopts this plugin and continues it's maintenance I'd be happy to chip in via patreon or something like that, I would take a crack at it, but I don't have the time or resources to learn programming at this given moment.

heinzight, 2024-10-03:
Help please...  I installed this but now I don't want it and there is no 'wrench' to get to be able to uninstall it. How do I uninstall it, please.

jneilliii, 2024-10-03:
Try http://octopi.local/recovery

heinzight, 2024-10-10:
Yes, that worked although I wasn't sure what or where to do the recovery. So it makes me wonder why this plug-in was not on the Blacklist if it's not maintained or at least the warning before not after install.

It seems that install/uninstall wiped out my login credentials (??) to the Pi 4B. I cannot get into the Pi from my computer. I took very careful notes in installing Octopi but it will not acept my username nor password. I can no longer see my files. I can't get into Octopi.org either (that one I do not have notes, long ago when I had a Pi3B.

heinzight, 2024-10-10:
Thank you Gina!

foosel, 2025-02-13:
Yeah... no. Sorry to say, but that just smells wrong with that name suffix and that empty an account.

ender-bit, 2025-02-27:
I will like to take ownership of the plugin, please let me know if you're interested in giving me access to the plugin

rpi10, 2025-04-12:
> ### Which Plugin?
> https://plugins.octoprint.org/plugins/touchui/
> 
> ### Link to ticket on plugin's issue tracker confirming abandonment
> [BillyBlaze/OctoPrint-TouchUI#535](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/535)
> 
> ### Is there someone open to adopting it?
> _No response_

I would like to take back the project. Your work has been truly amazing: THANK YOU!

ender-bit, 2025-04-13:
To touch ui

ender-bit, 2025-04-13:
To Touch UI, thats the plugin I would like to take ownership of

DanielmGM, 2025-08-09:
🙌 Someone is making a progress in this? i am willing to help mantain this project alive as it's the best of this kind

cameleonnzl, 2025-08-25:
So the last push on this was a little over four years ago. Do people really like it enough to want it kept alive? I had it in my download list and it's warned as abandoned.

CmdrOverBeard, 2026-10-13:
Hello OctoPrint community,

I would like to offer continued community maintenance for the abandoned
TouchUI plugin.

The maintained fork is called **TouchUI - ReVersiond**:

https://github.com/CmdrOverBeard/OctoPrint-TouchUI-ReVersiond

The original project, copyright notices, license and Git history have been
preserved. TouchUI was originally created by Paul de Vries / BillyBlaze.
This is a community-maintained continuation and not a claim of original
authorship.

The current release is:

https://github.com/CmdrOverBeard/OctoPrint-TouchUI-ReVersiond/releases/tag/v0.9.0

Changes made so far include:

- compatibility work for current OctoPrint installations
- support for the current webcam container and streaming methods
- fixes for the temperature graph layout
- a Raspberry Pi OS Bookworm HDMI kiosk assistant
- improved touchscreen scrolling and operation
- an extensible theme/template system
- Modern, LCARS-inspired, OctoDash-inspired and digital-rain themes
- updated installation and theme-development documentation

The plugin is currently being tested on OctoPrint 1.11 with Raspberry Pi OS
Bookworm. I am willing to maintain the project, review bug reports and continue
adapting it to future OctoPrint releases.

Before submitting a plugin repository update, I would appreciate guidance on
the preferred adoption process and whether the existing TouchUI plugin entry
should be transferred or whether TouchUI - ReVersiond should receive a separate
entry.

Maintainer: CmdrOverBeard  
Affiliation: IT Services Segbart

P.s.: Sorry my English is not the Yellow from The Egg . Text ist KI Translatet

jneilliii, 2026-10-13:
I noticed you don't have any historical github activity and curious if you developed your changes extensively using AI or not. If that is the case, then we would probably want to register your plugin separately rather than as an adoption.

I've scanned your plugin source using the [octoscanner](https://github.com/jacopotediosi/octoscanner) tool. Please find the results in the attached txt file. We would want to have these addressed prior to adoption/new registration. 

[octoscanner_results.txt](https://github.com/user-attachments/files/33393596/octoscanner_results.txt)

heinzight, 2026-10-14:
Jneillii,

Sorry, I really don't understand all this. I don't use or need TouchUI.

Tom

roblandry, 2026-10-15:
4 years later and nothing done. I opened this to inform people. Closing to stop receiving notifications as I doubt this will ever be revived.

jacopotediosi, 2026-10-15:
@roblandry please keep this open. You can stop notifications clicking the "unsubscribe" button in this page.
About the "I doubt this will ever be revived"... I wouldn't be so sure about that 😜

varek3d, 2026-11-08:
Okay, I still get the notification about this I guess because I had it.  I used claude to fix it. :-)  I suggest everyone do the same.  ENJOY!