OctoPrint/plugins.octoprint.org #1137: [Abandoned Plugin] SlicerSettingsTab

larsjuhw, 2022-12-11:
### Which Plugin?

https://plugins.octoprint.org/plugins/SlicerSettingsTab/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/tjjfvi/OctoPrint-SlicerSettingsTab/issues/5

### Is there someone open to adopting it?

@larsjuhw

jneilliii, 2022-12-11:
@tjjfvi giving you 14 days to respond prior to handing over maintenance of the plugin to larsjuhw and redirecting updates.

tjjfvi, 2022-12-11:
@jneilliii In a perfect world I'd continue to maintain those plugins, but I haven't, so... if someone else is willing to take it on, that's great; let me know if there's anything I can do to facilitate that

jneilliii, 2022-12-12:
Thanks for confirming. There's nothing necessary on your end for this thanks to the update check overlays.

jneilliii, 2022-12-12:
If @larsjuhw hasn't forked your repo yet it would be extremely smooth if you transfer ownership of the repo in it's settings @tjjfvi.

larsjuhw, 2022-12-12:
@jneilliii I haven’t forked it yet no.

tjjfvi, 2022-12-12:
> larsjuhw already has a repository in the tjjfvi/OctoPrint-SlicerSettingsTab network  

@larsjuhw Looks like you forked a fork of my repo, so I can't transfer it.

tjjfvi, 2022-12-12:
I'm also willing to give you admin access on the original repo, if that's useful.

larsjuhw, 2022-12-12:
> > larsjuhw already has a repository in the tjjfvi/OctoPrint-SlicerSettingsTab network
> 
> @larsjuhw Looks like you forked a fork of my repo, so I can't transfer it.

Forgot about that. Just deleted it, does it work now?

larsjuhw, 2022-12-12:
@tjjfvi thanks for the help!

tjjfvi, 2022-12-12:
Do you also want to take over <https://github.com/tjjfvi/OctoPrint-SlicerSettingsParser>? SlicerSettingsTab depends on it.

larsjuhw, 2022-12-12:
> Do you also want to take over https://github.com/tjjfvi/OctoPrint-SlicerSettingsParser? SlicerSettingsTab depends on it.

I already adopted it: https://github.com/OctoPrint/plugins.octoprint.org/pull/1136 :)

jneilliii, 2022-12-12:
you won't need to do the notices or update check overlay, just the plugin listing page.

closed 2022-12-15 by jneilliii