OctoPrint/plugins.octoprint.org #1141: [Abandoned Plugin] octoprint-grbl-plugin

synman, 2022-12-27:
### Which Plugin?

https://plugins.octoprint.org/plugins/octoprint-grbl-plugin/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/mic159/octoprint-grbl-plugin/issues/11

### Is there someone open to adopting it?

@synman

synman, 2022-12-27:
If feasible, I would like to repoint this plugin to the Better Grbl Support plugin (this plugin was its inspiration) and add @mic159 as an author in recognition of his efforts.

jneilliii, 2022-12-28:
I'm not opposed to doing something like that, but would be interested in hearing what other @OctoPrint/plugin-repository-maintainers think, and of course see if there's a response from @mic159.