OctoPrint/plugins.octoprint.org #1148: [Abandoned Plugin] Filament Sensor Simplified

valemaio2, 2023-01-19:
### Which Plugin?

https://plugins.octoprint.org/plugins/filamentsensorsimplified/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/LuckyX182/Filament_sensor_simplified/issues/71

### Is there someone open to adopting it?

@valemaio2

valemaio2, 2023-01-19:
I know the issue asking if the plugin has been abandoned has been opened for only five days, but there are other issues with the plugin opened back in October 2022 that are yet to receive a reply.

jneilliii, 2023-01-26:
@LuckyX182 any input? @derekantrican I see you have a pending PR, would you be interested in adopting if we don't hear back or possibly you and @valemaio2 combine efforts? Potentially just getting you both contributor rights to the current repo would be a good solution.

derekantrican, 2023-01-26:
@jneilliii I don't mind having contributor access, but I don't really want the responsibility of ownership here. I think @valemaio2 as the creator of this issue indicated they wanted to take over ownership

valemaio2, 2023-01-26:
@jneilliii I don't mind having ownership transferred if @derekantrican doesn't want to
 Of course, the more contributors the better.

valemaio2, 2023-02-01:
@jneilliii Thank you. Since this is my first public repo on Github, could you please expand on the changing the software update hook? Or even just link me a resource I can read upon?

derekantrican, 2023-02-01:
Similar to https://github.com/smartin015/continuousprint/pull/5/files (from the issue @jneilliii linked). To edit in GitHub, go to the repo and press the period (.) key. Find and replace any instances of "LuckyX182" with "valemaio2". Also reference the above linked issue for changes like author, bumping version, etc.

Let  me know if you need help (I literally just did this today for another plugin 😄 )

valemaio2, 2023-02-02:
@jneilliii I think I've just done it? I'm still trying to figure out how Github works, sorry. I've been looking at the issue you and derek linked, and did what I could based on that. @derekantrican I've invited you as a collaborator, feel free to accept or decline, as I believe you said you don't mind being a contributor. If you accept, I'll give you proper credit on the plugin homepage, and wherever else is needed.

jneilliii, 2023-02-03:
Not quite. I closed your PR because it's not complete. I'll assist you with the necessary changes later tonight when I get home after work.

derekantrican, 2023-02-27:
Has there been any update on getting this migration completed? Let me know if I can assist

jneilliii, 2023-02-28:
Sorry, dropped the ball on this one. I just looked at @valemaio2 's fork and it needs to have issues enabled and release created for version 0.3.4 before any overlays/adjustments to the plugin listing are done.