OctoPrint/plugins.octoprint.org #1149: [Listing Error] GCODE System Commands

moffitj, 2023-01-19:
### Which Plugin?

https://github.com/kantlivelong/OctoPrint-GCodeSystemCommands

### What kind of listing error are you reporting?

incorrect compatibility information on the plugin listing

### What specifically is broken or wrong?

Listing:    Python: >=2.7,<3
plugin states:    __plugin_pythoncompat__ = ">=2.7,<4"
https://github.com/kantlivelong/OctoPrint-GCodeSystemCommands/blob/b525ba6b0fa93fd6edc4eaa13e478a4e09a2480d/octoprint_gcodesystemcommands/__init__.py#L142

cp2004, 2023-01-30:
The error in the build log is this:

```
gcodesystemcommands.md generated an exception: while constructing a mapping
  in "<unicode string>", line 2, column 1:
    layout: plugin
    ^ (line: 2)
found duplicate key "featuredimage" with value "/assets/img/plugins/gcodesystemcommands/gcodesystemcommands_settings.png" (original value: "/assets/img/plugins/gcodesystemcommands/gcodesystemcommands_settings.png")
  in "<unicode string>", line 20, column 1:
    featuredimage: /assets/img/plugi ... 
    ^ (line: 20)
```

cp2004, 2023-01-30:
Confirmed fixed with PR #1155

closed 2023-01-30 by cp2004