OctoPrint/plugins.octoprint.org #1156: [Abandoned Plugin] OctoPrint-Webhooks

derekantrican, 2023-02-01:
### Which Plugin?

https://plugins.octoprint.org/plugins/webhooks

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/2blane/OctoPrint-Webhooks/issues/38

### Is there someone open to adopting it?

@derekantrican

jneilliii, 2023-02-01:
@derekantrican since you already have a fork it's just a matter of updating the plugin listing page to point to yours and add you to the authors list. Make sure to change the software update hook to point to your repo and bump the version and release with the new version number. If you need any points, I'd look at this adoption PR as reference of what they changed as part of the process, and don't hesitate to ask any questions for clarification.

https://github.com/OctoPrint/plugins.octoprint.org/pull/1027

derekantrican, 2023-02-01:
Thanks! I'll make sure I move my pending changes to a different branch so the "master" mirrors the current state of the official repo.

Once I've updated the update hook (and anything else) do I just come back here and give you the "go ahead" to update the plugin listing page?

jneilliii, 2023-02-01:
Submit a PR similar to the one above referencing this issue.

closed 2023-02-01 by jneilliii

derekantrican, 2023-02-01:
Thanks! At what point can I expect this to show up on plugins.octoprint.org and the alert to users in their octoprint interface?

jneilliii, 2023-02-01:
It typically is within 24 hours, but that depends on the user's configured TTL for plugin update notices.

cp2004, 2023-02-02:
The `pip` field must also be specified in the overlays file here:
https://github.com/OctoPrint/plugins.octoprint.org/blob/05874cbcfc08929aa40e776a8870b5667ea6d18b/_data/update_check_overlays.yaml#L52-L53

Similar to this, it overrides the `pip` field in the SW update config, so it should look the same as what you have the SW update hook looking like at the moment.

https://github.com/OctoPrint/plugins.octoprint.org/blob/05874cbcfc08929aa40e776a8870b5667ea6d18b/_data/update_check_overlays.yaml#L45-L47

derekantrican, 2023-02-02:
Thank you! I'll make that change

jneilliii, 2023-02-02:
Ah, thanks for the clarification @cp2004. I thought updating the user alone was enough to handle the redirect.