OctoPrint/plugins.octoprint.org #1208: [Listing Error] OctoLight Home Assistant

hansvanlin, 2023-07-05:
### Which Plugin?

https://plugins.octoprint.org/plugins/octolightHA/

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

link to installer is broken. 
[image]

jneilliii, 2023-07-05:
@mark-bloom looks like you need to fix your install URL. Also, please enable issues on your plugin repo.

mark-bloom, 2023-07-06:
Issues have been enabled on the plugin repo and a pull request has been opened for the corrected download link in the .md file.

Thank you again for the commentary.

jneilliii, 2023-07-06:
Pull request merged, should resolve issue once pushed to the site shortly.

closed 2023-07-06 by jneilliii