OctoPrint/plugins.octoprint.org #1212: [Abandoned Plugin] Filament Manager

wizardkitty, 2023-07-10:
### Which Plugin?

https://plugins.octoprint.org/plugins/filamentmanager/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/OllisGit/OctoPrint-FilamentManager/issues/120

wizardkitty, 2023-07-10:
The plugin has sat abandoned for some time and issues are starting to creep in. It no longer installs or functions on some machines.

jneilliii, 2023-07-10:
@OllisGit are you done with plugin development?

jneilliii, 2023-07-24:
per https://github.com/OctoPrint/plugins.octoprint.org/issues/1213#issuecomment-1647442098 @OllisGit is still around for maintaining plugin, was just on break.

closed 2023-07-24 by jneilliii