OctoPrint/plugins.octoprint.org #1213: [Abandoned Plugin] Spool Manager

wizardkitty, 2023-07-10:
### Which Plugin?

https://plugins.octoprint.org/plugins/SpoolManager/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/OllisGit/OctoPrint-SpoolManager/issues/333

wizardkitty, 2023-07-10:
https://github.com/mdziekon/OctoPrint-SpoolManager is a working version so they may be willing to adopt it.

jneilliii, 2023-07-10:
@mdziekon any interest in adopting the plugin? @OllisGit are you done with plugin development?

mdziekon, 2023-07-11:
@jneilliii yes, my fork might still need some technical preparation for full "take over" (like fixing dev / pre-release workflows), but other than that it should be ready for registration into plugins repository.

jneilliii, 2023-07-11:
Great, whenever you're ready you can submit a PR here for the changes. There are a couple of adoption PRs you can take a look at for what needs to be modified as example, but I'm general it's update authors list, URLs, and the update overlays. Feel free to ask any questions related here or in discord in the dev-plugins channel.

Hottabych-, 2023-07-23:
Hi, 
I don't know the processes if s.o. is willing to take over the responsibility. So my question when and where can I see a status change? At the moment the label is 'abandoned' here, will that change or is the label tight with the name (&plugin) of the original supporter?
My personal thank to @mdziekon to take over the responsibility for these two plugins (& PrintJobHistory; correct?) from olli.

mdziekon, 2023-07-23:
@Hottabych- can't speak for sure about the process, but I assume that as soon as I create a Pull Request here and it gets accepted & merged, the plugins repository should notify people about the new version. I guess @jneilliii could verify whether that's the case.

And yes, I'm planning to provide at least basic maintenance PrintJobHistory as well. It's taking me a bit longer than anticipated to prepare for SpoolManager take-over, but once I get past personal stuff, and prepare 2.0 version, I'll create a PR here to make things official.

jneilliii, 2023-07-23:
Yes, once we go through the process, end users will just get notified of a new version available from your forked version based on the software update overlay. So prior to submitting that PR, make sure you've done a version bump on your repo that is higher than the original. You can find a complete example adoption PR [here](https://github.com/OctoPrint/plugins.octoprint.org/pull/651/files).

OllisGit, 2023-07-24:
Hi! I AM STILL HERE ;-)

I know it was really quiet around my plugins, but I did a "sabbatical". So, no plugin-development, main focus was working in my new job.

I am happy that someone will take over the responsiblilty for the plugins....but...I don't want to hand over responsibility just yet.

So, my suggestion is that I get in contact with @mdziekon, review the changes in his fork and will discuss the next steps, (maybe here in the ticket)

flaviut, 2023-11-05:
Hey @OllisGit, I appreciate your hard work on these plugins; they've been very useful!

But despite your intent, it doesn't seem like you've gotten a chance to work on them. Can you please allow them to be formally abandoned so someone else can pick them up? There are some talented developers looking to maintain them.

dimo414, 2023-11-05:
@jneilliii IMO these tickets should be reopened, there's been no activity on these repos in months and no sign of @OllisGit.

panatale1, 2023-11-29:
Is this project still abandoned? I would gladly help adopt it

Webguyatwork, 2023-11-30:
I also created a fork and added external database support, tested with 2 separate installations of Octoprint on 2 separate printers. I can't say it couldn't benefit from some additional code review from other devs, but it does seem to be working with an external MySQL database for now.

See https://github.com/Webguyatwork/OctoPrint-SpoolManager if you'd like to play around with it in an offline dev environment. If @OllisGit will accept a PR, I can send one. It seems other PRs have gone unanswered.

broderickhyman, 2023-12-01:
@dojohnso I noticed that you are now the maintainer of PrintJobHistory, any interest in taking over this project as well? I am interested myself but don't think I have the time commitment at the moment to take on a full existing project. I hope to contribute some bug fixes to these projects though when possible.

dojohnso, 2023-12-01:
> @dojohnso I noticed that you are now the maintainer of PrintJobHistory, any interest in taking over this project as well? I am interested myself but don't think I have the time commitment at the [...]

if no one else wants to take it, i could. but like with PrintJobHistory, i'll only have time to focus on immediate bug fixes and minor enhancements, so if someone else wants to do more with it sooner, please feel free to step in. the process to take it is pretty straight forward, my PR is here as a reference if anyone is curious about what needs to be changed here. https://github.com/OctoPrint/plugins.octoprint.org/pull/1248

Webguyatwork, 2023-12-01:
I really want to see my external database enhancements released to everyone, but I really don't want to take over ownership of the plugin due to my own time constraints. If we can decide here who can take it over officially, I would definitely like to help contribute.

Hottabych-, 2023-12-03:
Sounds like a good fit for me, we have 4 persons who are willing to contribute: @dojohnso, @Webguyatwork, @broderickhyman and @panatale1 so no one is alone. 
Now we need a person with the hat on ... @dojohnso? you have the experience with the last one from Olli
If anybody need I can help in testing; but progamming ... ;-)  you don't need Assembler, Pascal and Fortran 77 any more.

dojohnso, 2023-12-03:
ha sure, i'll volunteer as tribute :) i'll work on this tomorrow

dojohnso, 2023-12-06:
submitted https://github.com/OctoPrint/plugins.octoprint.org/pull/1255

Hottabych-, 2023-12-09:
As I understand this this plugin is now under maintenance of @dojohnso - correct? So why is there still an 'abandoned' sign? Second question, @Webguyatwork could you merge your database enhancements into the dev tree of dojohnso? I think this would be a great feature for people who have more than one printer.

Webguyatwork, 2023-12-09:
> > As I understand this this plugin is now under maintenance of @dojohnso - correct? Thanks a lot for this. So why is there still an 'abandoned' sign? Second question, @Webguyatwork could you merge your database enhancements into the dev tree of dojohnso? I think this would be a great feature for people who have more than one printer.
> 

I'm waiting on the official hand-off to complete, then yes, I'll attempt to merge my changes in.

closed 2023-12-09 by jneilliii

Webguyatwork, 2023-12-11:
Pull request for external DB support is pending approval - https://github.com/dojohnso/OctoPrint-SpoolManager/pull/17

PR #1258 (2023-12-09): adopt SpoolManager
resolves #1213