OctoPrint/plugins.octoprint.org #1222: [Abandoned Plugin] PrintJobHistory

dojohnso, 2023-08-02:
### Which Plugin?

https://github.com/OllisGit/OctoPrint-PrintJobHistory

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/216

### Is there someone open to adopting it?

@dojohnso

jneilliii, 2023-08-02:
@OllisGit has reported recently on https://github.com/OctoPrint/plugins.octoprint.org/issues/1213#issuecomment-1647442098 that he is back from his plugin sabbatical. Maybe he can enlighten us on when he'll get around to your specific PR.

dojohnso, 2023-08-02:
> @OllisGit has reported recently on [#1213 (comment)](https://github.com/OctoPrint/plugins.octoprint.org/issues/1213#issuecomment-1647442098) that he is back from his plugin sabbatical. Maybe he can enlighten us on when he'll get around to your specific PR.

ok awesome, sounds good! totally understand that reasoning of his sabbatical. @OllisGit let me know if there's anything I can do to help!

cp2004, 2023-08-02:
I'd also add that generally any PR or issue that hasn't had a response doesn't count as an issue confirming abandonment - we'd need an issue opened specifically asking if the plugin is abandoned, then once *that* receives no response, it could be considered abandoned.

dojohnso, 2023-08-23:
> I'd also add that generally any PR or issue that hasn't had a response doesn't count as an issue confirming abandonment - we'd need an issue opened specifically asking if the plugin is abandoned, then once _that_ receives no response, it could be considered abandoned here.

https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/223

How long before it counts? This has been about 3 weeks plus the 3 months for the PR for the fix that everyone commenting on the project wants, plus the 1 year and 5 months since the last update to the repo.... sorry for the snarky tone, it's just frustrating when it's this hard to get a simple change made to a popular plugin that clearly affects the majority of people using it. If there is a reason he's not approving that change, a response of any kind explaining why would be nice and very acceptable, but he hasn't been active on that plugin in a minute.

dojohnso, 2023-10-05:
> I'd also add that generally any PR or issue that hasn't had a response doesn't count as an issue confirming abandonment - we'd need an issue opened specifically asking if the plugin is abandoned, then once _that_ receives no response, it could be considered abandoned here.

bump

Filmman586, 2023-11-26:
When will This be fixed, kinda frustrating that there is a fix and it can’t be implemented because the creator can’t be reached.

foosel, 2023-11-27:
I've just marked this officially as abandoned, so if someone wants to take over maintenance now, we can go ahead. I hate doing this, but this has been going on like this for far too long now and frustrating way too many people.

dojohnso, 2023-11-27:
> I've just marked this officially as abandoned, so if someone wants to take over maintenance now, we can go ahead. I hate doing this, but this has been going on like this for far too long now and frustrating way too many people.

Thank you Gina!! I would be happy to take this over at least for now, to get these initial fixes out that people have been waiting for. Just let me know what I need to do.

foosel, 2023-11-27:
Quoting the one and only @jneilliii:

>If you want to adopt the plugin it's just a matter of submitting a PR here that configures the correct update check overlays, the plugin listing data adding you as an author and a bumped version with release on your fork of the plugin. Here is an example of a completed adoption PR to see what I mean.
>
>https://github.com/OctoPrint/plugins.octoprint.org/pull/1157/files

closed 2023-11-27 by jneilliii