OctoPrint/plugins.octoprint.org #1235: [Suspicious Plugin Activity] OctoPrint-BedCheckAI

hn3000, 2023-10-09:
### Which Plugin?

https://plugins.octoprint.org/plugins/bedcheckai/

### What kind of suspicious activity are you reporting?

data processing not mentioned in the privacy policy

### Link to a ticket on the plugin's issue tracker about the problem

https://github.com/printpal-io/OctoPrint-BedCheckAI/issues/2

### Further information you want to provide

(as discussed on mastodon https://berlin.social/@octoprint@fosstodon.org/111205857101877608 )

printpal-io, 2023-10-09:
We have responded to this issue at: https://github.com/printpal-io/OctoPrint-BedCheckAI/issues/2

Both points brought up by the user are already implemented in the listing bio and the plugin tab, very clearly. 

Not sure why this issue was opened in the first place.

jneilliii, 2023-10-09:
> very clearly.

```
The ML model is large so it is run on the cloud. 
```

Maybe clarifying the type of data that is submitted to the cloud would make sense here, i.e. mention the fact that snapshots are sent to the cloud.

jneilliii, 2023-10-09:
> When the BedCheck Macro/button is triggered, the current camera snapshot is sent to the cloud for ML inference and the various pixel values are returned.

added to plugin listing to make it more obvious that snapshots are sent to the cloud.

closed 2023-10-09 by jneilliii