OctoPrint/plugins.octoprint.org #1240: [Abandoned Plugin] OctoPrint-Enclosure

gavinhelyar, 2023-11-03:
### Which Plugin?

https://plugins.octoprint.org/plugins/enclosure/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/README.md

aakatz3, 2024-06-10:
I'm working on trying to familiarize myself with the code. I'd be willing to maintain the plugin on a best-effort basis, but I am a full-time student. A lab I work in needs this plugin to be functional, so I know I can devote time into updating it to make it functional again, and I can handle pull requests from others, but I may be very slow with new features.

jneilliii, 2026-10-08:
@jacopotediosi there are older closed issues referencing this as well, but this was the latest open one. I'll flag the plugin abandoned now that you've mentioned the repo was archived.

jacopotediosi, 2026-10-08:
Ops, I missed this one for some reason.

Copying from the duplicated issue I opened (#1496):

> @vitormhenrique archived the repository in March 2026, after a long period of inactivity during which only the most urgent PRs were merged.
> 
> I use this plugin quite a lot and had already made some changes in my fork to fix some crashes.
> In particular, the plugin's code is currently rather messy, and since everything runs in a single thread, any crash is also a fire hazard imo, as it can leave the enclosure heater or other components turned on (already happened many times to me).
> 
> I'd like to adopt this plugin. My plan is to do a heavy refactor and rewrite large parts of the code. It will take a few weeks or months, so please wait for me.

jacopotediosi, 2026-10-08:
@jneilliii maybe we should remove the following, to avoid a duplicated popup:

https://github.com/OctoPrint/plugins.octoprint.org/blob/c6a795d209059dc3730000206bf2257d33de3384/_data/notices.yaml#L281-L286

[image]

jneilliii, 2026-10-08:
Forgot that was even in there. Done.

jasenjkrueger-arch, 2026-11-01:
Hello,

My Name is Jasen and I 've been using Raspberry Pi, Octoprint, OctoPod and the Enclosure Plugin for many years.  I would like this plugin to stick around and there are things I wish I could fix or even make better.  I have done a lot of different programming and I am interested in working on it / maintaining it.  I have some Arduino, Python, VBA, VBS and various Power app programming but I don't know how these plug-ins work in general but I am willing to look at it and see if it is something I can do.  Where do I start? What programing language / software do I need?

jacopotediosi, 2026-11-01:
@jasenjkrueger-arch please wait, I'm already adopting it and it is under a heavy refactor in my fork. I think it will be ready in a month.