OctoPrint/plugins.octoprint.org #1265: [Listing Error] Octoprint-LCD-HD44780

haggis-supper, 2023-12-29:
### Which Plugin?

https://git.kunsmann.eu/kunsi/Octoprint-LCD-HD44780

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

Server not found for the link.

jneilliii, 2023-12-29:
@Kunsi is this something that you'll be fixing or should I just go ahead and remove the plugin from the repo?

jneilliii, 2023-12-29:
@haggis-supper I see that kunsi has a mirror on GitHub, but don't know if that's been kept up to date or not.

https://github.com/Kunsi/Octoprint-LCD-HD44780

Kunsi, 2023-12-29:
oh damn, i migrated to git.franzi.business a couple months ago.

Apparently, the plugin does not work with recent versions of OctoPrint and since i don't use octoprint currently and i thus i have no incentive of fixing it, i'd recommend removing the listing, i will re-add the plugin when i get back to 3D printing.

closed 2023-12-30 by jneilliii