OctoPrint/plugins.octoprint.org #1268: [ async_from_user_password() missing 1 required positional argument: 'api_base_url' ] PSU Control - Meross

dennislia, 2024-01-03:
### Which Plugin?

https://github.com/VRGhost/OctoPrint-PSUControl-Meross

### What kind of listing error are you reporting?

incorrect compatibility information on the plugin listing

### What specifically is broken or wrong?

1/3/2024
trying to "Test Login" to Meross Cloud I get the following error message in the PSU Control-Meross Plug-in, using the same user name and password that I use to login to the Meross App on my iphone:
" async_from_user_password() missing 1 required positional argument: 'api_base_url'  "
why?

jneilliii, 2024-01-03:
This issue tracker is reserved for issues with the plugin repository, for issues with the phone themselves please open issue on their repository. I've linked it here for your convenience.

https://github.com/VRGhost/OctoPrint-PSUControl-Meross/issues

closed 2024-01-03 by jneilliii

dennislia, 2024-01-03:
Sorry for the inconvenience  … the same problem is reported there … following the conversations it appears to have been fixed … any idea if the fix is available or how to get ?

Thanks, Denny

jneilliii, 2024-01-03:
copy/paste this in plugin manager > get more > ...from URL.
```
https://github.com/luizbon/OctoPrint-PSUControl-Meross/archive/master.zip
```

dennislia, 2024-01-03:
Brilliant … this added API URL pulldown … I selected US and was able to login with my Meross App username and password … my meross smart devices were populated into the Device list …
Thanks again, Denny

> On Jan 3, 2024, at 3:31 PM, jneilliii ***@***.***> wrote:
> 
> https://github.com/luizbon/OctoPrint-PSUControl-Meross/archive/master.zip