OctoPrint/plugins.octoprint.org #1307: [Listing Error] OctoPrint-3DGeeks

foosel, 2024-06-12:
### Which Plugin?

https://plugins.octoprint.org/plugins/geeks3d/

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

Plugin home page errors out. It's unclear whether https://www.3dgeeks.xyz/ is the new page. Reviews of the Android app suggest it no longer works correctly. Last commit 4 years ago, so this might also be abandoned.

foosel, 2024-06-12:
@2BWorks can you comment on the above and if possible provide a working home page link and also one for the privacy policy of your service?

jacopotediosi, 2025-04-14:
Also the Android application, of which this Octoprint plugin is a companion plugin, is no longer available on the Play Store: https://play.google.com/store/apps/details?id=work.twob.third

commit dca141e8 (2025-05-14): chore: remove geeks3d

Website and app are gone, author doesn't respond, let's get rid of this
for good.

Closes #1307
closed 2025-05-14 by foosel