OctoPrint/plugins.octoprint.org #1319: [Abandoned Plugin] SpoolManager

rtrout, 2024-09-04:
### Which Plugin?

https://github.com/dojohnso/OctoPrint-SpoolManager

### Link to ticket on plugin's issue tracker confirming abandonment

https://community.octoprint.org/t/is-spool-manager-still-in-development/60025/6

jneilliii, 2024-09-04:
would like to get @dojohnso 's input directly rather than taking information second hand.

rtrout, 2024-09-05:
> would like to get @dojohnso 's input directly rather than taking information second hand.

Doug has now responded here:
https://github.com/dojohnso/OctoPrint-SpoolManager/issues/54#issuecomment-2330254412

WildRikku, 2024-09-05:
I saw mdziekon/Octoprint-SpoolManager which was another fork of Olli's repository. It has quite a few commits but no recent ones. I posted an issue there asking if the maintainer is still interested in the plugin. Otherwise, I might think about taking it over. I was originally interested in FilamentManager for an integration with another plugin but saw that it has been abandoned, too, and people were using SpoolManager now.

dojohnso, 2024-09-05:
> would like to get @dojohnso 's input directly rather than taking information second hand.

Yeah I don't have the time to work on it so you all can move ahead with whatever is needed to take this over. same for my PrintJobHistory plugin

WildRikku, 2024-09-11:
Alright, I can take over the SpoolManager plugin. I spent the last night figuring out some things and think I can handle the codebase and existing issues and also have some fun. :)

I created a fork, activated issues and made some notes already. https://github.com/WildRikku/OctoPrint-SpoolManager

I also forked this repository and started preparing the pull request that would seal the takeover if I understood it correctly. I use this PR as reference: https://github.com/OctoPrint/plugins.octoprint.org/pull/1255/files
I am unsure about the changes made to `notices.yaml` and `update_check_overlays.yaml`. The first I think is to re-distribute the plugin without making a release? I'd write a bugfix I want to work on anyway and make a release as an exercise, so no change necessary then?
And for `update_check_overlays.yaml`, that looks like it does the same? Or do they work together and one is for notification and one is for actually pointing to new source? Would I add my lines to the files or replace those from the previous PR?

jneilliii, 2024-09-12:
> Or do they work together and one is for notification and one is for actually pointing to new source? Would I add my lines to the files or replace those from the previous PR?

This is exactly the situation, they each control different aspects of the notification and update process. 

> I also forked this repository and started preparing the pull request that would seal the takeover if I understood it correctly. I use this PR as reference: [#1255 (files)](https://github.com/OctoPrint/plugins.octoprint.org/pull/1255/files)

Yes, this is a good example. You would also want to remove the abandoned setting currently in the top of SpoolManager.md file.

WildRikku, 2024-09-12:
Thank you. I have opened #1324 accordingly (at least I hope so).

rtrout, 2024-09-15:
Big thank you @WildRikku, 1.7.4 fixed the CSV download bug for me.

jacopotediosi, 2025-04-14:
Since the plugin has been adopted, shouldn't this issue be closed?

closed 2025-04-14 by jneilliii