OctoPrint/plugins.octoprint.org #1322: PrintJob History Abandoned

jneilliii, 2024-09-08:
Yeah I don't have the time to work on it so you all can move ahead with whatever is needed to take this over. same for my PrintJobHistory plugin

_Originally posted by @dojohnso in https://github.com/OctoPrint/plugins.octoprint.org/issues/1319#issuecomment-2332344915_

KevinVidomski, 2024-12-03:
I have 20+ years experience in software development including web technologies.  I have a lot of time on my hands right now.  I'm going to look through the two plugins and see if this is something I could handle.

vojtakaniok, 2025-02-18:
I created [PR](https://github.com/dojohnso/OctoPrint-PrintJobHistory/pull/31) with fix for one [issue](https://github.com/dojohnso/OctoPrint-PrintJobHistory/issues/23).
Let me know if this plugin is owned by someone once again or I can maintain it for a while.
@jneilliii

jneilliii, 2025-02-18:
If you'd like to adopt the plugin @vojtakaniok, it's just a matter of submitting a PR similar to the previous adoption PR that dojohnso did [here](https://github.com/OctoPrint/plugins.octoprint.org/pull/1248/files).

vojtakaniok, 2025-02-18:
Thanks for answer !
I will do some cleanup in the repo and pubslih the PR for adopting.

Larry2368, 2025-02-23:
I would be willing to help when I can.  My time is often limited.  I am new
to using GitHub and I am not sure what language the plug-in is written in,
but I have been coding since the late '70s.

vojtakaniok, 2025-02-23:
@Larry2368
Thanks for your willingness to help!  You can checkout my repo and help me to maintain it !
https://github.com/vojtakaniok/OctoPrint-PrintJobHistory

jneilliii, 2025-02-24:
Thanks @vojtakaniok adoption PR has been merged.

closed 2025-02-24 by jneilliii

Kjubyte, 2025-08-26:
@vojtakaniok Do you still actively maintain the project? The disabled issue tracker makes me believe you have abandoned it.

vojtakaniok, 2025-08-26:
@Kjubyte Yes, I'm still maintaining the project. The issues was disabled by default when forking the repo. 
Fixed