OctoPrint/plugins.octoprint.org #1328: [Listing Error] OctoPrint-GitFiles

jacopotediosi, 2024-10-01:
### Which Plugin?

https://plugins.octoprint.org/plugins/gitfiles/

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

Both homepage and source are 404

jacopotediosi, 2024-10-01:
@OutsourcedGuru idk if you can take a look

jneilliii, 2024-10-01:
I will be marking this one as abandoned. OutsourceGuru has been gone for some time. I did find this fork of his repo though on github.

https://github.com/itnachos/OctoPrint-GitFiles

jneilliii, 2024-10-01:
There are also other forks that have had updates to them since they were split. I know @tidelinio was actively developing the plugin to work for his workflow. 

https://github.com/tidelineio/OctoPrint-GitFiles

jneilliii, 2024-10-01:
plugin listing removed since it has been taken down from GitHub by original author.

closed 2024-10-01 by jneilliii