OctoPrint/plugins.octoprint.org #1334: [Abandoned Plugin] CostEstimation

Hillshum, 2024-11-17:
### Which Plugin?

https://plugins.octoprint.org/plugins/costestimation/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/OllisGit/OctoPrint-CostEstimation/issues/27

### Is there someone open to adopting it?

@hillshum

Hillshum, 2024-11-18:
Also [my PR for this plugin](https://github.com/OllisGit/OctoPrint-CostEstimation/pull/26) is now going on 3 months with no response.

jneilliii, 2024-11-18:
Please file a PR for taking over the plugin. It would be similar to the previous one submitted by Olli when he took it over, changes linked below. Bump the version on your fork and create a tag and release that matches that version number. 

https://github.com/OctoPrint/plugins.octoprint.org/pull/630/files

However, instead of `author:` with a comma separated list, you should use `authors:` as a list. 

```
authors:
- Hilton Shumway (or however you want your name to show)
- Olli
- Sven Lohrmann
```

closed 2024-11-21 by jneilliii