OctoPrint/plugins.octoprint.org #1342: [Abandoned Plugin] OctoPrint-PrettyGCode

ryanschulze, 2025-01-05:
### Which Plugin?

https://plugins.octoprint.org/plugins/prettygcode/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/Kragrathea/OctoPrint-PrettyGCode/issues/165

cyberman, 2025-01-07:
Hello everyone,

I am interested in taking over this project. At the moment, I know little about GITHUB projects, but I'm sure I can make this work for everyone's benefit. The code is a bit older, but it shouldn't be hard to update the code. Just replacing jQuery with modern vanilla JS should pay off. The current three.js also offers significant added value.

eliliam, 2025-03-24:
> I am interested in taking over this project. 
@cyberman I see you have a fork of it in your GitHub profile, however it's identical to the original upstream repository. Are you planning on updating it and start to maintain your fork of it? I'd love to see some work done there; if you don't intend to maintain it, I would edit your comment as it comes across as you stating that you will be the new maintainer.

pythonhub1, 2025-05-02:
I would like to take over this project. I know about github and I do a lot of coding. I also 3d print a lot with OctoPrint.

jneilliii, 2025-05-02:
I see that @Kragrathea has merged in some of the older pull requests last month, just awaiting a new version bump and release. Makes me wonder if this is abandoned, if additional contributors should be added to the repo, or transfer of ownership?

rudraforweb, 2025-07-18:
If someone has taken over the project, would it be possible to remove the "This plugin is not available" message?
Thanks!

leandro-costa-oliveira, 2025-09-02:
@ryanschulze I can help. :)

Quantenkosmos, 2025-12-27:
If you want, you can try https://github.com/Quantenkosmos/OctoPrint-PrettyGCode - a fork with help of gemini. For me it works. Please use it, and take over ishumilin.