OctoPrint/plugins.octoprint.org #1357: [Listing Error] OctoPrint-OctoLabel

jacopotediosi, 2025-03-28:
### Which Plugin?

https://plugins.octoprint.org/plugins/octolabel/

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

https://github.com/LowieGoossens/octolabel/archive/master.zip returns the following error, preventing the plugin installation via Plugin Manager: `the given path has multiple possibilities: #<Git::Ref:0x00007ff2c7ba2d10>, #<Git::Ref:0x00007ff2c7ba2a90>`

jacopotediosi, 2025-03-28:
@LowieGoossens ?

jneilliii, 2025-03-28:
That is strange, never seen that before. The following URL should work, but normally GitHub will direct what is used in the plugin listing normally. 

```
https://github.com/LowieGoossens/octolabel/archive/refs/heads/master.zip
```

kantlivelong, 2025-03-28:
For whatever reason there is a `master` tag which should be removed.

LowieGoossens, 2025-03-28:
Thank you for notifying me, i will solve this :)

LowieGoossens, 2025-03-28:
https://github.com/LowieGoossens/octolabel/archive/refs/tags/master.zip

LowieGoossens, 2025-03-28:
I removed the master tag and cleaned up the releases, it seems to work now.

closed 2025-03-28 by jneilliii