OctoPrint/plugins.octoprint.org #1366: [Listing Error] Octoprint Nexus AI

jacopotediosi, 2025-04-21:
### Which Plugin?

https://plugins.octoprint.org/plugins/nexus_ai/

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

The main link in the listing, https://fiber-punk.com/pages/nexusai, is no longer reachable:

[image]

jacopotediosi, 2025-04-21:
@fiber-punk fiberpunk1

jneilliii, 2025-04-21:
seems the whole domain is down with same error. wonder if project is dead and this plugin should be removed or not. if we don't hear back from @fiber-punk @fiberpunk1 in 30 days I think that's what we'll do.