OctoPrint/plugins.octoprint.org #1371: [Listing Error] ATXPiHat

jacopotediosi, 2025-05-03:
### Which Plugin?

https://plugins.octoprint.org/plugins/atxpihat/

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

The plugin’s listed homepage URL, https://www.baprojectworkshop.com/, is no longer resolvable:
[image]

Additionally, the listing includes another link labeled baprojectworkshop.com, which actually points to https://wp.me/p98gmw-7g. That URL redirects to https://baprojectworkshop.wordpress.com/, which is also no longer reachable:
[image]

My hypothesis is that the custom domain baprojectworkshop.com was once mapped to the WordPress site. After the WordPress site was shut down, the custom domain may have expired and is likely no longer under the plugin maintainer’s control.

jacopotediosi, 2025-05-03:
Cc (the plugin owner): @banichow

jneilliii, 2025-05-03:
If we don't hear back I vote for removal, since the plugin is useless without the hat.

jacopotediosi, 2025-05-03:
According to https://github.com/banichow/OctoPrint-Atxpihat/issues/1, the hardware was supposed to be open source, but the schematics were on the site which is no longer reachable.
With the last push in 2018, the incompatibility for python3, the project site down, and only one active instance in the last month according to the stats, I would also opt for removal.

jacopotediosi, 2025-05-03:
By browsing the site using web.archive.org, I found this post stating that the project was no longer maintained as early as 2018:
[image]

closed 2025-05-03 by jneilliii