OctoPrint/plugins.octoprint.org #1373: [Abandoned Plugin] Ikea Tradfri

jacopotediosi, 2025-05-07:
### Which Plugin?

https://plugins.octoprint.org/plugins/ikea_tradfri/

### Link to ticket on plugin's issue tracker confirming abandonment

screenshot of a twitter chat below

jacopotediosi, 2025-05-07:
I had a quick chat with the owner @ralmn on Twitter, via the link in the description on his GitHub profile. He confirmed that the plugin is abandoned.

[image]

ralmn, 2025-05-07:
I confim. I don't have any octoprint instance (moving to Prusa MK4) and no more Tradfri Hub, and no enough time to maintain this plugin.